CadQuery/CQ-editor #420: Permit disabling fov change during reload

BlueDrink9, 2024-01-11:
When I'm changing scaling to check something is changing, or figuring out how someone else's model works by tweaking things, every time the model reloads, my FoV is reset, making it impossible to objectively assess what is changing. Being able to fix the fov across reloads would help a lot, similar to openscad's behaviour

#204 would also probably help this.

adam-urbanczyk, 2024-01-11:
AFAICT it is supported:
[image]

BlueDrink9, 2024-01-12:
Thank you Adam! It's a bit unclear what that option refers to initially. Would you welcome a PR with updated descriptions for those options? How terse do you want those text boxes to be?

closed 2024-01-12 by BlueDrink9