CadQuery/CQ-editor #424: vi mode using vim-spyder

roel-v, 2024-01-15:
I've been toying with ways to add vi-style editing support to the editor. To reduce duplication of work I've focused on re-using as much as possible from the vim-spyder plugin for Spyder. I've gotten a proof of concept mostly working, most of what remains is filling in boilerplate. Before I move ahead on this though I'd like to discuss how to go about it and whether it's worth investing my time (if there's no chance this will ever get merged anyway, I won't).

Getting this Spyder plugin to work with cq-editor is not difficult, the options revolve around the trade off between whether one wants to be able to use the plugin as-is, without any changes, which would require changes to the cq-editor internals; or a way that is less invasive but would require changes to the plugin, breaking compatibility, which is mostly an 'in the future' issue (and fwiw the last update to this plugin is from 6 months ago).

Roughly I see three paths:

- 'Full compatiblity' - this basically entails emulating Spyder's full plugin system in cq-editor. It would allow more plugins to be used as is (although I don't know of any, especially not ones relevant to cq-editor's use case, then again I'm not well versed in Spyder's ecosystem). It would require fairly significant changes to cq-editors internals as in Spyder, everything is a plugin (including things like the editor, debugger etc). Those changes constitute adding interfaces so that most/all parts of cq-editor can be accessed and modified by plugins from the outside. Code complexity think hundreds of lines over many/almost all of the code.

- 'Halfway' - with some minor modifications to both cq-editor's internals and (a fork of) the spyder-vim plugin, most of the functionality of the plugin could be unlocked. This is mostly adding the very basics of Spyder's plugin system to cq-editor (adding a plugin registry for external plugins, instantiating and initializing plugins, some basic callbacks/notifications) and some changes to the UI (basically, adding a QVBoxLayout to all docking windows that don't have it so that the UI can be customized from plugins). Code complexity think dozens/low hundreds of lines across a few files. This is most closely to what I have now in my prototype.

- Forget about plugins and just copy the required functionality from spyder-vim. This would require the same UI changes as above but the rest could be copied over from the plugin and integrated into cq-editor where needed. I think it would still be possible to keep some files verbatim from spyder-vim so that if they make any changes, it would only be a matter of copying over those files again to get the benefit of the upgrades. Complexity think a few dozen/low hundreds across a few files.

I would appreciate feedback on this matter, even if negative so that I know if I should continue thinking about this. Thanks.

adam-urbanczyk, 2024-01-18:
TBH I think this does not fit into the minimalistic scope of CQ-editor. Given that you can use any external editor including ([neo]vim), what is the reason to even consider this?

roel-v, 2024-01-23:
OK fair enough. The reason for a vim mode is to move cq-editor more in the direction of a full-fledged IDE. I use Openscad's gui now in the way you say, with one editor window and the gui as a preview basically, but it's quite hacky. Cq-editor has more functionality for which you'd prefer to not switch between applications even less (like using the debugger). IMO having a proper IDE that does more than just integrate a bare-bones text editor, preview and debugger is needed to make cadquery a serious option for designing 3d parts. There's room between Fusion 360-style 'point and click' design and a fully code-focused approach where the IDE is an afterthought. But if the vision for cq-editor is to be a lightweight, bare-bones editor-with-debugger, then OK, I can take that into account.

adam-urbanczyk, 2024-02-01:
You might want to consider developing a Spyder plugin for CQ/OCP. Then you'll get all the goodies of spyder. It did cross my mind, but I will not work on this myself in the foreseeable future.

roel-v, 2024-02-05:
Yes I looked at that, but that's not as straight forward as it looked at first, either. And I'm not keen on taking on a whole project to maintain by myself. But that would be the way to go for anyone wanting extended functionality.