CadQuery/CQ-editor #425: Request: new stable release - or clarify

DoppioT, 2024-01-26:
Hello!

First of all, I'd like to thank the contributors of CQ-editor, it is really a great project!

As a new user of CQ-editor, the main indications I see when looking into installing the software is:

- the recommended method is via pre-built packages, and
- there's a stable release 

If I install the stable release and try to follow the Cad query examples, I'll soon realize that the stable package ships with a version of CadQuery (2.1) that doesn´t support a lot of major features (sketches and assemblies just to mention the first two that come to mind). 

This point was already raised in other issues - see #274 , #306, #395 - but all the issues are a few months old at least

So the questions would be: should pre-built packages still be the recommended way to install CQ-editor?

- If so, can a new stable release be delivered?
- If not, I propose that the readme be updated to reflect that pre-built package are not the recommended way to install CQ-editor, and instead recommend Conda/Mamba. In this second case I'll open a PR for the update.

adam-urbanczyk, 2024-01-28:
Try this instead: https://github.com/CadQuery/CQ-editor#development-packages. Even if there is a release, no pyinstaller packages will be produced (see #395).

adam-urbanczyk, 2024-01-28:
I did update the readme for clarity.

DoppioT, 2024-01-28:
Thanks! That solves the issue for me.

closed 2024-01-28 by DoppioT

Fran5ois, 2024-12-04:
Hello here,

When I follow the [link](https://github.com/CadQuery/CQ-editor/releases/tag/nightly) given in the installation instructions - Development Packages, the windows version offered for download turns out to pack with CadQuery 2.0.

As a result, for example, the cylinder function isn't gere.

As a beginner it took me a while to figure that the error messages did not come from errors in my code... or did it?

Isn't CadQuery supposed to be at version 2.4? That's what the "check for updates" button tells me, but where can I find it?

CQ editor seems very promising, but has it been given up on? If there is such a better way to use CadQuery could someone please point me to it?

I really hope to find a way to use CQ editor in its full potential soon... 

Thank you in advance.

adam-urbanczyk, 2024-12-05:
AFAICT it packages the master version (from the time of building of the installer):

https://dev.azure.com/cadquery/CQ-editor/_build/results?buildId=3922&view=logs&j=8438bad5-9a3c-5a20-4304-a4804615829c&t=78b00efc-8d05-51a9-2f07-ca31c05f8748&l=471

Could you provide more details?

Fran5ois, 2024-12-05:
In the link you provide, I can see INFO:constructor.fcp:fetching: cadquery-master-2.4.0_28_g3cd327a.tar.bz2

2.4, that's the version I am after!

However when I reach "About" in CQ editor, here is what I get:

[image]

And here is what I get when I click "Check for CadQuery updates":

[image]

Also when I try the function cylinder it doesn't work (while box does):

[image]

Why do I have an old version if the installer while compiled with 2.4?

When I start CQ-editor I get this error message :

Namespace(filename=None)
TKOpenGl | Type: Error | ID: 0 | Severity: High | Message:
  Error! OpenGL context reports version 4.6 but does not export required functions for 4.5 (glGetnTexImage)
Please report this issue to OpenGL driver vendor 'intel'

Then CQ-editor starts anyway...

Could it be the reason why I get an older version ?

Also I am thinking, wouldn't it help if the installer filename mentioned the CadQuery version with which it was built?

Again, thank you very much for your help.