CadQuery/CQ-editor #426: Dark mode theme

yamsu, 2024-01-31:
I'm finding CQ-Editor really useful to learn and use cadquery. However, I couldn't figure out how to get a dark theme set for the entire app and not just the editor widget.

I have a crude work around that involve editing the CQ-editor python files directly
- for dark mode across the app
conda activate
python3.1 -m pip install 
[](./cq-editor/lib/python3.11/site-packages/cq_editor/__main__.py)
- for dark mode in the console 
  edit the following [[./cq-editor/lib/python3.11/site-packages/cq_editor/widgets/console.py]

jmwright, 2024-01-31:
Possibly related: https://github.com/CadQuery/CQ-editor/issues/304

yamsu, 2024-01-31:
> Possibly related: #304

There is also the fork mentioned in #304 that does this and allows one to choose the theme. Although pyqdarktheme has an option to detect the system colorscheme. But i have not tested this.
```
qdarktheme.setup_theme('auto')
```

closed 2024-01-31 by yamsu