CadQuery/CQ-editor #427: Is it possible to use cq-warehouse inside a Pre-Built Packages ?

healBvdb, 2024-02-03:
I would like to use some of the possibilities of cq-warehouse directly from a pre-built package. Is there a way to do that ?

jmwright, 2024-02-04:
I don't think you can with the PyInstaller packages, or at least not without a lot of trial and error. Conda is probably your best bet if you want to customize the installation. There are instructions in the CadQuery documentation on how to install conda without touching your local Python environment.

https://cadquery.readthedocs.io/en/latest/installation.html#install-the-conda-package-manager

adam-urbanczyk, 2024-02-04:
[Nightly](https://github.com/CadQuery/CQ-editor/releases/tag/nightly) uses conda constructor, so it should be possible.

healBvdb, 2024-02-05:
Thanks for your answers… I finally followed @jmwright suggestion and installed everything using CadQuery documentation.

closed 2024-02-05 by healBvdb