CadQuery/CQ-editor #429: When exception is raised, the "Current traceback" window is empty

zwn, 2024-02-16:
See result when raising `RuntimeError('example').

[image]

I need to run the script outside of cq-editor to get the stack trace in console.

I am on Debian, installed cq-editor with micromamba:

```
cq-editor                             master           0.2_60_g7d74cf7        cadquery 
```

lifeisafractal, 2024-03-05:
I'm seeing the same issue. I'm on Ubuntu 23.10 using miniforge:
```
cq-editor                 master          0.2_61_g03df6fa    cadquery
```

smurfix, 2024-04-08:
Quick fix for this:

```diff
diff --git cq_editor/widgets/traceback_viewer.py cq_editor/widgets/traceback_viewer.py
index d5e0baa..4b1043d 100644
--- cq_editor/widgets/traceback_viewer.py
+++ cq_editor/widgets/traceback_viewer.py
@@ -56,6 +56,6 @@ class TracebackPane(QWidget,ComponentMixin):
             root = self.tree.root
             code = code.splitlines()
-            tb = [t for t in extract_tb(tb) if '<string>' in t.filename] #ignore highest frames (debug, exec)
+            tb = extract_tb(tb)

             for el in tb:
                 #workaround of the traceback module
```

smurfix, 2024-04-08:
I opened PR #431 with a better fix that doesn't suppress tracebacks in called modules.

They're too helpful for finding the actual problem …

PR #431 (2024-06-04): Show the traceback.
Also trace into external modules.

Closes #429.
commit 98711d8a (2024-06-04): Show the traceback. (#431)

* Show the traceback.

Also trace into external modules.

Closes #429.

* Use itertools for clarity

* Cleanup

* Remove empty line

* Cover new functionality with tests

---------
closed 2024-06-04 by adam-urbanczyk