CadQuery/CQ-editor #432: fit and render on file opening

aavogt, 2024-04-12:
`cq-editor file.py` should run `file.py` without having to press F5. This change would make it more like freecad.

[cq-editor loads the file with this code](https://github.com/CadQuery/CQ-editor/blob/03df6fa64beda77e374bf4160411e36e0764d332/cq_editor/main_window.py#L64). I changed it to also do what F5 does:

```python
     if filename:
            self.components['editor'].load_from_file(filename)
            self.components['editor'].triggerRerender.emit(True)
```
With the above change, `cq-editor file.py` renders file.py without user intervention, but the object is too small. I suspect the fit (Shift-F1) runs too early. Here's a screenshot:

[image]

adam-urbanczyk, 2024-04-12:
Regarding the first point: no thanks. Regarding the second point:
[image]

closed 2024-04-12 by adam-urbanczyk

aavogt, 2024-04-12:
I have that checked. If I `touch file.py` 10 seconds after starting cq-editor I get the right result.