CadQuery/CQ-editor #434: Installation with Micromamba fails

jscatena88, 2024-04-24:
When trying to install I get the following:
```
jason@jason-ThinkPad-T14s-Gen-4:~/micromamba$ micromamba create -n cadquery
Empty environment created at prefix: /home/jason/micromamba/envs/cadquery
jason@jason-ThinkPad-T14s-Gen-4:~/micromamba$ micromamba activate cadquery 
(cadquery) jason@jason-ThinkPad-T14s-Gen-4:~/micromamba$ micromamba install -c cadquery cq-editor=master
conda-forge/linux-64                                        Using cache
conda-forge/noarch                                          Using cache
cadquery/noarch                                               No change
cadquery/linux-64                                             No change
nodefaults/noarch                                             No change
nodefaults/linux-64                                           No change
warning  libmamba Problem type not implemented SOLVER_RULE_STRICT_REPO_PRIORITY
warning  libmamba Problem type not implemented SOLVER_RULE_STRICT_REPO_PRIORITY
warning  libmamba Problem type not implemented SOLVER_RULE_STRICT_REPO_PRIORITY
warning  libmamba Problem type not implemented SOLVER_RULE_STRICT_REPO_PRIORITY
warning  libmamba Problem type not implemented SOLVER_RULE_STRICT_REPO_PRIORITY
warning  libmamba Problem type not implemented SOLVER_RULE_STRICT_REPO_PRIORITY
warning  libmamba Problem type not implemented SOLVER_RULE_STRICT_REPO_PRIORITY
warning  libmamba Problem type not implemented SOLVER_RULE_STRICT_REPO_PRIORITY
warning  libmamba Problem type not implemented SOLVER_RULE_STRICT_REPO_PRIORITY
warning  libmamba Problem type not implemented SOLVER_RULE_STRICT_REPO_PRIORITY
warning  libmamba Problem type not implemented SOLVER_RULE_STRICT_REPO_PRIORITY
warning  libmamba Problem type not implemented SOLVER_RULE_STRICT_REPO_PRIORITY
warning  libmamba Problem type not implemented SOLVER_RULE_STRICT_REPO_PRIORITY
warning  libmamba Problem type not implemented SOLVER_RULE_STRICT_REPO_PRIORITY
error    libmamba Could not solve for environment specs
    The following package could not be installed
    └─ cq-editor master**  is not installable because it requires
       └─ cadquery master.* , which requires
          └─ ocp 7.7.2.* , which conflicts with any installable versions previously reported.
critical libmamba Could not solve for environment specs
(cadquery) jason@jason-ThinkPad-T14s-Gen-4:~/micromamba$ 
```

It seems there is some dependency mismatch. From the repository I can see that there is no `7.7.2` available, only `7.7.0`: https://anaconda.org/CadQuery/ocp

adam-urbanczyk, 2024-04-24:
ocp 7.7.2 is available here https://anaconda.org/conda-forge/ocp I don't understand the error to be honest.

jscatena88, 2024-04-26:
Does a fresh install work for you?

adam-urbanczyk, 2024-04-26:
I can only try on win right now, and this does work: `micromamba.exe env create -n _cq_test -c cadquery cq-editor=master`

jscatena88, 2024-04-26:
Alright I figured it out the fix was setting the "channel-priority" config in micromamba. After running `micromamba config set channel_priority flexible` it worked as expected

jscatena88, 2024-04-26:
I am not very familiar with Conda in general, but it seems as long as an older/incompatible version of `ocp` is offered in the same channel as `cq-editor` installation with strict ordering will be impossible. And strict ordering is the default when you install `micromamba`.  I'm not 100% sure though

adam-urbanczyk, 2024-04-28:
Good catch, I'm going to remove those builds.

jscatena88, 2024-04-30:
Let me know if you want me to test installation with strict ordering after you make the change. Also feel free to close this issue if you want, or I can close it after testing the changes, whichever you prefer

adam-urbanczyk, 2024-04-30:
Could you try with the initial settings again? I think I solved the issue, but would be nice to have a confirmation.

jscatena88, 2024-04-30:
Yep that seems to work even with strict ordering for me. Closing the issue

closed 2024-04-30 by jscatena88