CadQuery/CQ-editor #435: Can't launch CQ-editor nor cadquery on Raspberry pi

nadirion, 2024-05-05:
Hello, smart people,
I have an issue, when I launch cq-editor or import cadquery in python I get same error:

(base) pi@raspberrypi:~ $ cq-editor
qt.qpa.plugin: Could not find the Qt platform plugin "wayland" in ""
Traceback (most recent call last):
  File "/home/pi/mambaforge/bin/cq-editor", line 6, in <module>
    from cq_editor.__main__ import main
  File "/home/pi/mambaforge/lib/python3.10/site-packages/cq_editor/__main__.py", line 12, in <module>
    from .main_window import MainWindow
  File "/home/pi/mambaforge/lib/python3.10/site-packages/cq_editor/main_window.py", line 5, in <module>
    import cadquery as cq
  File "/home/pi/mambaforge/lib/python3.10/site-packages/cadquery/__init__.py", line 22, in <module>
    from .occ_impl import exporters
  File "/home/pi/mambaforge/lib/python3.10/site-packages/cadquery/occ_impl/exporters/__init__.py", line 18, in <module>
    from .dxf import exportDXF, DxfDocument
  File "/home/pi/mambaforge/lib/python3.10/site-packages/cadquery/occ_impl/exporters/dxf.py", line 6, in <module>
    from ezdxf import units , zoom
ImportError: cannot import name 'zoom' from 'ezdxf' (/home/pi/mambaforge/lib/python3.10/site-packages/ezdxf/__init__.py)

Does anyone know, what can I do? I believe "wayland" issue and ezdxf issue are separate, but I could be wrong. Thanks.

jmwright, 2024-05-06:
How did you install CQ-editor? I don't think the Raspberry Pi is officially supported.

nadirion, 2024-05-06:
I followed instructions on https://cadquery.readthedocs.io/en/latest/installation.html

# Install to ~/mambaforge
curl -L -o mambaforge.sh "https://github.com/conda-forge/miniforge/releases/latest/download/Mambaforge-$(uname)-$(uname -m).sh"
bash mambaforge.sh -b -p $HOME/mambaforge

# Activate
source $HOME/mambaforge/bin/activate

Then:

conda create -n cq
conda activate cq
mamba install cadquery

When that didn't work, tried:

mamba install -c cadquery -c conda-forge cq-editor=master

MWStonton, 2024-07-06:
I think as you say - you have 2 issues. I have the "graphics" issue with crostini (Linux on ChromeOS on an x86 machine) and don't know how to fix it but I've also had the ezdxf issue on ARM Linux in a VM on a Mac and have managed to solve it.
I've posted the following as stonto1 here https://www.reddit.com/r/cadquery/comments/1duyjiv/installing_cadquery_on_apple_silicon/

"Not sure if this might help in macOS and I can't quite provide step by step instructions but I got cadquery running in a Linux VM on Apple silicon (having followed the instructions to install via conda) after getting dependency issues in python (think the report when trying a basic setup - import cadquery - was related to zoom from ezdxf).
What I think this issue was is that the version of ezdxf for ARM stored on conda (anaconda.org/conda-forge/ezdxf) was v0.14.2 vs 1.1.0 for x86. (The icons on this page still show the old version being the one for macOS so hopefully this is still a fix...)
I went to github.com/mozman/ezdxf and downloaded the source zip (under the green code icon). I think I grabbed the latest but there is a selection box to the left where you can choose stable.
Back in python you can use pip install (in the conda "environment" where cadquery is installed) to install source by pointing it to the unzipped source download. This updates the ezdxf package to the later version which was all that was broken in my case.
I did this in a Linux VM but it's quite possible it's the same issue in macOS. apologies for the sketchy instructions but I did this a while ago and meant to revisit but haven't as yet. Maybe if this works for you, you could post a more precise set of instructions"

I have a question to pass back - where did you find an ARM version of CQ-editor to install on the Raspi?

nadirion, 2024-07-07:
Thank you, Mark, I'll try your solution and will let you know.
As for installing on Raspberry, installation through conda automatically
does it for aarch64 architecture. Follow these commands
under Addind Cq-Editor to an Existing Environment here:
https://cadquery.readthedocs.io/en/latest/installation.html#adding-a-nicer-gui-via-cq-editor

Nadir.