CadQuery/CQ-editor #439: CQ Editor cannot find current project

lenianiva, 2024-06-15:
Consider a project setup like this:
```
/
|- pyproject.toml
|- poetry.lock
|- mylib
    |- __init__.py
    |- mystery.py
|- example
    |- example.py
```
where
```python
# mystery.py

def mystery():
    return Cq.Workplane().box(10, 5, 5)
```
in a `poetry shell` environment, this succeeds:
```
(mylib-3.12) [example/] $ python3 -c "import mylib.mystery"
```
but if `example.py` contains any reference to `import mylib.mystery`, loading it up in `cq-editor` will fail, unless the script starts with this:
```python
# example.py
import os, sys, os.path

root = os.path.dirname(os.getcwd())
print(root)

sys.path.append(root)

import mylib.mystery
```
The spooky thing is after this part of the script is loaded into cq-editor, commenting it out would make the effect persist. Is there a way for cq-editor to recognize the current venv so it can load libraries referenced by cad scripts?

lenianiva, 2024-06-15:
Addendum: This seems to be a problem with `sys.path`

The `sys.path` outside of cq-editor is
```
(mylib-3.12) [example/] python3 -c "import sys; print(sys.path)"
['', '/usr/lib/python312.zip', '/usr/lib/python3.12', '/usr/lib/python3.12/lib-dynload', '/home/aniva/.cache/pypoetry/virtualenvs/nhf-Q4XoAPRK-py3.12/lib/python3.12/site-packages', '/home/aniva/Projects/mylib']
```
The `sys.path` inside cq-editor is
```
['/opt/cq-editor-bin/base_library.zip', '/opt/cq-editor-bin', '/opt/cq-editor-bin/IPython/extensions']
```

adam-urbanczyk, 2024-06-16:
You seem to be using the packaged CQ-editor, so it has no relation with your venv. Does using `preferences->debugger->Add script dir to path` solve your issue?

lenianiva, 2024-06-18:
> You seem to be using the packaged CQ-editor, so it has no relation with your venv. Does using `preferences->debugger->Add script dir to path` solve your issue?
> 
> Reading your question for the second time, what I mention above won't help. For now you could install your module in the (conda) env shipped with the installer.

I found a workaround which is to put a sentinel file at the project root like this:
```python
import mylib.mystery

obj = mylib.mystery.mystery()
show_object(obj)
```
but when I made changes to the object, cq-editor cannot pick it up even if I click Render. This is not a major problem since I can just copy whatever object I'm working on to this sentinel file.

adam-urbanczyk, 2024-06-19:
So I think the conclusion here is that CQ-editor has no "ope dir" or "open project" functionality. Might be a nice thing to add.