CadQuery/CQ-editor #440: Where did Linux nightly release go?

Wren6991, 2024-06-16:
Hi, first of all thank you for all your hard work on CadQuery and the editor. I have been installing CQ-editor on my machines by following the directions here: https://cadquery.readthedocs.io/en/latest/installation.html

```bash
curl -LO https://github.com/CadQuery/CQ-editor/releases/download/nightly/CQ-editor-master-Linux-x86_64.sh
sh CQ-editor-master-Linux-x86_64.sh
```

I tried to install on a new machine today, and I just got a 200-byte file which says:

```
<?xml version="1.0" encoding="utf-8"?><Error><Code>BlobNotFound</Code><Message>The specified blob does not exist.
RequestId:06da1206-d01e-005b-21ab-bfd86e000000
Time:2024-06-16T05:09:35.9682405Z</Message></Error>
```

If I go to the "Nightly" release tag then indeed that file is now missing (there is just Windows/MacOS):

https://github.com/CadQuery/CQ-editor/releases/tag/nightly

Where did it go? I think this has changed within the last few days. If the Linux build was broken in some way, is it possible to have the last known-good version added to a separate release?

Thanks!

adam-urbanczyk, 2024-06-16:
There was a glitch in the ci pipeline, should be fine now

Wren6991, 2024-06-16:
Yes it's back, thanks!

closed 2024-06-16 by Wren6991