CadQuery/CQ-editor #441: show_object fails with TypeError

lorenzncode, 2024-06-21:
This results in error (latest commit).   No error with previous commit.
```py
from cadquery.occ_impl.shapes import segment, vertex

e1 = segment((0, 0, 0), (10, 10, 10))
p1 = e1.positionAt(0.5)
v1 = vertex(p1)

show_object(v1)
```
[image]

lorenzncode, 2024-06-21:
Potential fix - return `NotImplemented` on comparison with non-Vector?

https://github.com/lorenzncode/cadquery/blob/688683f9e1de777b3e95224dd94c312da17ee195/cadquery/occ_impl/geom.py#L232-L236

adam-urbanczyk, 2024-06-22:
Thanks, actually it should be return False

adam-urbanczyk, 2024-06-22:
Solved by https://github.com/CadQuery/cadquery/pull/1610

closed 2024-06-22 by adam-urbanczyk