CadQuery/CQ-editor #442: Not runable after installing according to Readme

TDB-ev, 2024-06-21:
After installing cq-editor using:
```
mamba install -c cadquery -c conda-forge cq-editor=master
```
it is not possible to run `cq-editor` and yields an error:
```...
...lib/python3.11/site-packages/nptyping/typing_.py", line 51, in <module>
    Bool8 = np.bool8
...
AttributeError: module 'numpy' has no attribute 'bool8'. Did you mean: 'bool'?
```

Causes by some depency problem (nptype / numpy related).

After downgrading `numpy` it does work:

```
mamba install numpy=1.26.8
```

justbuchanan, 2024-06-23:
+1 I had the same issue and used the same workaround (downgrading numpy to something before 2.0).

It looks like there's a proposed fix for `nptyping` in https://github.com/ramonhagenaars/nptyping/pull/114.

justbuchanan, 2024-06-23:
Attempted fix in https://github.com/CadQuery/cadquery/pull/1612

GiulioBj, 2024-09-02:
Hi, I have the same problem. Do you raccomand to downgrade cadquery from 2.4.0 version to another older version? If I downgrade only numpy there is no compatibility between the 2 libraries.

adam-urbanczyk, 2024-09-03:
Does installing cadquery from the cadquery channel work: `mamba install -c conda-forge cadquery::cadquery=master`? In the meantime we should start releasing 2.5.

jmwright, 2025-02-06:
This should be fixed by the latest release of CadQuery.

closed 2025-02-06 by jmwright