CadQuery/CQ-editor #446: Path.abspath not working, use Path.absolute

btesanovic, 2024-08-09:
Running on OSX Arm M2 , with latest master release and python 3.10.6 
I had to replace in Editor and Widget from Path.abspath() to Path.absolute() for GUI to load file and render it

DanielJoyce, 2024-08-10:
Yep, just saw this on linux as well...

Try to run the example script and I get an error just above the log:

'Path' object has no attribute 'abspath'

DanielJoyce, 2024-08-10:
So if you use PathLib, its absolute(), but if you use os.path its abspath.

Yay more fun with dynamic languages

DanielJoyce, 2024-08-10:
In my case I had to change line 195 of debugger.py from abspath to absolute.

lorenzncode, 2024-08-10:
I noticed this today too with a new install.
```
(cqdev) PS>micromamba list path
List of packages in environment: "E:\\tools\\micromamba_envs\\envs\\cqdev"

  Name      Version  Build            Channel
---------------------------------------------------
  path      17.0.0   py311h1ea47a8_0  conda-forge
  pathspec  0.12.1   pyhd8ed1ab_0     conda-forge
```
https://path.readthedocs.io/en/latest/history.html#v17-0-0

version 16.15.0 working
(cqdev) PS>micromamba install -c conda-forge path=16

closed 2024-08-11 by adam-urbanczyk

PaulKGrimes, 2025-01-02:
This error still exists in a fresh install of version 0.3.0dev on Windows via conda

PR #447 (2024-08-11): Abspath fix
Fixes #446