CadQuery/CQ-editor #453: Selection invalidation on mouse drag PATCH

neri-engineering, 2024-10-02:
I have a very simple proposed change for cq-editor.

Here is the hypothetical scenario.

I open up cq-editor with some object that takes up the center of the viewing area, hit F5 to make it visible.
Next, I left-click the mouse from the bottom left corner of the viewing area, which is a space filled with void.  I then drag the mouse up and to the right, and the object rotates as expected.
Lastly I let go of the left mouse button ('release') when the mouse cursor is DIRECTLY OVER my part.  This causes the part to be selected.

I find this selecting behavior to be annoying.  I implemented a fix which only selects the object if the mouse didn't move a whole lot during the mouse drag operation.

Patch file link: https://sourceforge.net/p/nl10/code/HEAD/tree/patches/cq-editor/select-invalidation-on-drag.patch

Please note, this bug tracker WILL NOT allow me to upload a PATCH file, even though its claim is that such an attachment IS SUPPORTED.

adam-urbanczyk, 2024-10-03:
You are right! I did not notice before, but now it is quite annoying.

neri-engineering, 2024-10-03:
Thanks for the confirmation.  I'm motivated to start my "winter projects".

adam-urbanczyk, 2024-10-10:
I added a fix to #452 , so stay tuned.

closed 2025-03-01 by jmwright