CadQuery/CQ-editor #455: AttributeError: 'Editor' object has no attribute 'run_cell_action'

crankycyclops, 2024-10-04:
I followed these steps to get cadquery and CQ-Editor installed:

```bash
python3 -m venv ~/local/cadquery
source ~/local/cadquery/bin/activate
pip3 install --upgrade pip
pip3 install cadquery

pip3 install PyQt5 spyder pyqtgraph logbook
pip3 install git+https://github.com/CadQuery/CQ-editor.git
```

This pulled in the latest CQ-editor commit from master. I'm now getting the following error when I try to run the cq-editor command:

```
(cadquery) james@sauron ~/local $ cq-editor
Warning: None of the following fonts is installed: ['']
Traceback (most recent call last):
  File "/home/james/local/cadquery/bin/cq-editor", line 8, in <module>
    sys.exit(main())
  File "/home/james/local/cadquery/lib/python3.10/site-packages/cq_editor/__main__.py", line 21, in main
    win = MainWindow(filename=args.filename if args.filename else None)
  File "/home/james/local/cadquery/lib/python3.10/site-packages/cq_editor/main_window.py", line 44, in __init__
    self.prepare_panes()
  File "/home/james/local/cadquery/lib/python3.10/site-packages/cq_editor/main_window.py", line 89, in prepare_panes
    Editor(self),
  File "/home/james/local/cadquery/lib/python3.10/site-packages/cq_editor/widgets/editor.py", line 89, in __init__
    self._fixContextMenu()
  File "/home/james/local/cadquery/lib/python3.10/site-packages/cq_editor/widgets/editor.py", line 107, in _fixContextMenu
    menu.removeAction(self.run_cell_action)
AttributeError: 'Editor' object has no attribute 'run_cell_action'
```

crankycyclops, 2024-10-04:
I can get CQ-Editor to run if I manually open `~/local/cadquery/lib/python3.10/site-packages/cq_editor/widgets/editor.py` and comment out lines 107-110:

```python
#menu.removeAction(self.run_cell_action)
#menu.removeAction(self.run_cell_and_advance_action)
#menu.removeAction(self.run_selection_action)
#menu.removeAction(self.re_run_last_cell_action)
```

adam-urbanczyk, 2024-10-11:
Here is the env specification: https://github.com/CadQuery/CQ-editor/blob/master/cqgui_env.yml You used your own installation method, does the spyder version match what is in this yaml?

crankycyclops, 2024-10-11:
> Here is the env specification: https://github.com/CadQuery/CQ-editor/blob/master/cqgui_env.yml You used your own installation method, does the spyder version match what is in this yaml?

I used the pip installation method mentioned in the documentation: https://cadquery.readthedocs.io/en/latest/installation.html

It looks like that pulled in Spyder 6, which is a version above what's in the yml file.

davidedelvento, 2024-11-23:
In case anybody lands here and feel lost, a temporary workaround is to run the following (while being in the activated pip virtual environment)

```
pip install spyder==5
```

closed 2025-02-06 by jmwright