CadQuery/CQ-editor #457: Transparency alpha option for show object appears inverted

CasperH2O, 2024-10-30:
The show_object alpha property level appears to work "inverted" between CQ-Editor (0.3.0dev) and OCP CAD Viewer (2.5.3).

Best I can tell, if I compare with HTML/CSS standards, an alpha/opacity value of 1 should be not transparent, and 0 is fully transparent. See here for RGBA: https://www.w3schools.com/css/css_colors_rgb.asp

Currently, CQ-Editor shows fully transparent with a value 1.0, which is opposite from what I would expect.

Code snippet, comment out ocp_vscode when running in CQ-Editor:

```
import cadquery as cq
#from ocp_vscode import *

result1 = cq.Workplane("front").box(2.0, 2.0, 0.5)
result2 = cq.Workplane("bottom").box(2.0, 2.0, 0.5)

show_object(result1, name="Box Alpha 1.0", options={"alpha": 1.0})
show_object(result2, name="Box Alpha 0.1", options={"alpha": 0.0})
```

Result in CQ-Editor:

[image]

Result in OCP CAD Viewer:

[image]

closed 2025-03-02 by jmwright

CasperH2O, 2025-03-02:
Thank you Jeremy!

jmwright, 2025-03-02:
@CasperH2O Thanks for reporting it.

PR #499 (2025-03-02): Fixed alpha setting being backwards in show_object
Closes #457 

The core issue was that alpha was being passed to a `SetTransparency` method, and the values of alpha and transparency are inverted relative to each other. This fix compensates for that.

Some other issues came up while trying to fix this bug. The current code does not allow for the `show_object` options and the color/alpha GUI controls in the object settings panel to be kept properly in sync. The alpha from `show_object` was being overridden sometimes by the GUI control, and the color control was not reflecting what color the ob [... cut]