CadQuery/CQ-editor #458: Unable to Compile CadQuery-Based CQ-Editor Scripts with PyInstaller Spec File on Windows

PanagiotisMenounos, 2024-11-13:
### Description

Hello,

I'm using CQ-Editor to experiment with various designs and workflows. After finalizing my designs, I aim to transition into pure Python to create scripts that can be automated and perform more advanced tasks.

#### Minimal Python Script

Here's a minimal Python script I'm working with:

```python
import cadquery as cq

result = cq.Workplane("front").box(2.0, 2.0, 0.5)
result.val().exportStl("box.stl")
```

#### PyInstaller Spec File
When I attempt to compile this script using PyInstaller, here's the basic .spec file generated after running `pyinstaller main.py`:

```python
# -*- mode: python ; coding: utf-8 -*-

a = Analysis(
    ['main.py'],
    pathex=[],
    binaries=[],
    datas=[],
    hiddenimports=[],
    hookspath=[],
    hooksconfig={},
    runtime_hooks=[],
    excludes=[],
    noarchive=False,
    optimize=0,
)
pyz = PYZ(a.pure)

exe = EXE(
    pyz,
    a.scripts,
    [],
    exclude_binaries=True,
    name='main',
    debug=False,
    bootloader_ignore_signals=False,
    strip=False,
    upx=True,
    console=True,
    disable_windowed_traceback=False,
    argv_emulation=False,
    target_arch=None,
    codesign_identity=None,
    entitlements_file=None,
)

coll = COLLECT(
    exe,
    a.binaries,
    a.datas,
    strip=False,
    upx=True,
    upx_exclude=[],
    name='main',
)
```

#### Issue Encountered
After running the generated .exe file, I encounter a series of errors related to missing dependencies such as OCP,casadi._casadi, missing DLLs, etc.

Since CQ-Editor provides binary releases, I believe this is the best place to seek assistance on how to compile for Windows using PyInstaller.

#### Request for Assistance
- **Any suggestions on what the correct `.spec` file should look like to successfully compile simple pure python cadquery for Windows?**

Thank you for your help!

jmwright, 2024-11-13:
@PanagiotisMenounos You could take a look at this fork, which at least at one time built successfully on Windows. 

https://github.com/jmwright/CQ-editor/blob/master/.github/workflows/pyinstaller-builds-actions.yml

PanagiotisMenounos, 2024-11-19:
Thanks for your assistance, @jmwright ! Since you've already set everything up, the first step I took was to compile the CQ-Editor for Windows.   

## CQ-Editor Compilation - **GitHub Actions** - (Windows Fix)

FYI Here is the final file that successfully built the CQ-Editor using **GitHub Actions**:
  
**pyinstaller-builds-actions.yml**

```yml
name: build
on:
  schedule:
    - cron: '0 0 * * 1'
  workflow_dispatch:
    inputs:
      type:
        description: 'Whether to build a single file (onefile) or directory (dir) dist'
        required: true
        default: 'dir'
jobs:
  build-windows:
    runs-on: windows-2019
    steps:
    - uses: actions/checkout@v4
    - uses: conda-incubator/setup-miniconda@v3
      with:
        miniconda-version: "latest"
        channels: conda-forge,defaults
        auto-update-conda: true
        python-version: 3.9
        activate-environment: test
    - name: Install CadQuery and pyinstaller
      shell: powershell
      run: |
        conda install python=3.9
        conda install -c cadquery -c conda-forge cq-editor=master cadquery=master ipython=8.4.0 jedi=0.17.2 pyqtgraph=0.12.4
        pip install --upgrade pyinstaller==5.13.0
        pip install pipwin
        pipwin install numpy
        pip install path
        pip install git+https://github.com/gumyr/cq_warehouse.git#egg=cq_warehouse
        pip install git+https://github.com/meadiode/cq_gears.git@main
        pip install -e "git+https://github.com/CadQuery/cadquery-plugins.git#egg=cq_cache&subdirectory=plugins/cq_cache"
    - name: Run build
      shell: powershell
      run: |
        conda info
        pyinstaller pyinstaller.spec ${{ github.event.inputs.type }}
        $opensslPath = Get-Command openssl | Select-Object -ExpandProperty Source
        $opensslDir = Split-Path -Parent $opensslPath
        Copy-Item "$opensslDir\libssl-1_1-x64.dll" "D:\a\CQ-editor\CQ-editor\dist\CQ-editor\" -ErrorAction SilentlyContinue
        Copy-Item "$opensslDir\libcrypto-1_1-x64.dll" "D:\a\CQ-editor\CQ-editor\dist\CQ-editor\" -ErrorAction SilentlyContinue
        Copy-Item D:\a\CQ-editor\CQ-editor\pyinstaller\CQ-editor.cmd D:\a\CQ-editor\CQ-editor\dist\
    - uses: actions/upload-artifact@v4
      with:
        name: CQ-editor-Windows
        path: dist
```
## Python CadQuery Compilation - **GitHub Actions**

After some experimentation, I determined the necessary requirements and actions needed to successfully compile a basic CadQuery example with **GitHub Actions**:

### Actions Overview

1. **Conda Install**:
   - Install Python 3.9
   - Add channels: `cadquery` and `conda-forge`
   - Install CadQuery from the master branch
2. **Pip Install**:
   - Upgrade PyInstaller to version 5.13.0
[... 113 lines cut ...]
          upx=True,
          console=True)

exclude = ()
a.binaries = TOC([x for x in a.binaries if not x[0].startswith(exclude)])

coll = COLLECT(exe,
               a.binaries,
               a.zipfiles,
               a.datas,
               strip=False,
               upx=True,
               name='CQ-editor')

```

As you can see, I've removed many components, including runtime hooks, and the compiled version still works. Should I be concerned about potentially overlooking something important?

## Python CadQuery Compilation - **Local venv**

I aim to compile the project using a **local virtual environment (venv)** instead of **conda**. However, I'm encountering issues locating the `OCP.cp39-win_amd64.pyd` file and the necessary DLLs. 

Any recommendations?

Thank you in advance

closed 2024-11-21 by PanagiotisMenounos