CadQuery/CQ-editor #466: show_object works only in the CQ-editor not in imported files

DeBalog, 2024-12-26:
I have a file in CQ Editor containing the invoker code:

```python
import cadquery as cq
import os
import sys

lib_path = os.path.abspath("./")  # relativer Pfad zu einem absoluten Pfad umwandeln
if lib_path not in sys.path:
    sys.path.append(lib_path)

from show_test import *
```

which is just importing show_test.py.

show_test.py contains:

```python
import cadquery as cq
show_object (cq.Workplane ("XY").box(30,20,10))
```

Running the invoker in CQ-Editor brings up the following error:
[image]
NameError: name 'show_object' is not defined

The same code works fine if it is directly in CQ-Editor.

This is a big nuisance because I need show_object for debugging and my project is spread across multiple files.

Is there a simple solution to use show_object in a file which is invoked from CQ-Editor but which is not in CQ-Editor directly?

I use 
[image]
on Windows 11.

jmwright, 2024-12-29:
Hi @DeBalog 

Here is one way to solve the issue (pass the `show_object` function around):
```python
# Script running inside CQ-editor
from second_module import make_box

make_box(show_object)
```

```python
# Imported module ("second_module")
import cadquery as cq

def make_box(show_object):
    print(globals())
    result = cq.Workplane().box(10, 10, 10)

    show_object(result)
```

I also tried injecting `show_object` into `globals()` which works in some contexts, but does not seem to in this one.

DeBalog, 2024-12-29:
Hi @jmwright, thanks for you solution, I give it a try and report here.

DeBalog, 2025-03-14:
Thank to @jmwright recommendation, I've used show_object frequently in other modules. I've found it helpful to implement the set_show_object(so) function for each module. Once called, show_object can be used throughout the entire module. This works well. I've attached the code in case others have similar problems.

```python
show_object = None
def set_show_object (f):
    global show_object
    show_object = f

# Store the function
set_show_object (so) # lokal
```

closed 2025-03-14 by DeBalog