CadQuery/CQ-editor #467: Feature-Requests in CQ-Editor

DeBalog, 2024-12-26:
I used CQ-Editor a lot and now I tend to use an external editor. To enhance CQ-Editor I recommend:
* multiple tabs for opening multiple files. This would be a great benefit for all users.
* The ouput of print inside the CQ-Editor. It took me a week to find out, that the ouput was in a terminal somewhere in the background
[image]
* darkmode, because in the night the white background hurt my eyes if I switsch back to CQ-Editor from my dark mode editor
* Scales on the axes. It is disturbing, that a box of size (1,1,1) and (1000,1000,1000) are displayed equal in the editor
[image]
[image]
* a deleted breakpoint has no visual effect. Here is a breakpoint:
[image]
and here I just deleted it
[image]
did you see a difference? I didn't either. A visual feedback shoult help a lot. I clicked a lot to make the breakpoint vanish. Nothing helped.
* Sometimes CQ-Editor ignore my wish to save my code. I found some traces in the Log viewer:
[image]
* Line numbers are sometimes wrong / missleading:
the function definition is here between line Numbers 49/50:
[image]
By clicking the function definition it reveals its true line number:
[image]

I use 
[image]
on Windows 11.

jmwright, 2025-02-07:
It would be nice if all `print` statements, even from imported modules, output to the log viewer. For instance, I am developing a module and have imported it in CQ-editor to view the 3D result. I want to do print debugging in the imported module, but to do that I need to start CQ-editor form the terminal and check back to the terminal for my debugging messages. It would be great if those showed up in the log viewer.

I am not sure how to do that without redirecting stdout, and that comes with other issues I think.

jmwright, 2025-02-11:
As of #481 the result of `print` statements will now be appended to the log viewer. The encoding issue that you posted about should be fixed now too.

jmwright, 2025-02-12:
It's not a full dark theme mode yet, but in #484 the preferences drop downs were fixed so you can now select a darker color scheme for the editor (Edit->Preferences->Code Editor->Color scheme). I have mine set to `Monokai`, which gives a color scheme similar to the Sublime text editor's default, if you have ever used that. You can also change the background of the 3D viewer as well to make it darker (Edit->Preferences->3D Viewer->Background color). This makes about 60% of the window darker, which helps.

jmwright, 2025-02-12:
Everything in this issue has either been fixed recently, or is covered in a previous issue.

Dark theme: #304 
Scale markings on axes: #340 
Multiple tags: #65

closed 2025-02-12 by jmwright