CadQuery/CQ-editor #469: `AttributeError: module 'cadquery.occ_impl.exporters' has no attribute 'utils'`

jdegenstein, 2025-01-12:
CQ-editor fails to start because of a file deleted in CadQuery/cadquery:
```
[20:35:55.046432] ERROR: Uncaught exception occurred
Traceback (most recent call last):
  File "...lib/python3.12/site-packages/cq_editor/widgets/debugger.py", line 280, in render
    reload_cq()
  File "...lib/python3.12/site-packages/cq_editor/cq_utils.py", line 218, in reload_cq
    reload(cq.occ_impl.exporters.utils)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'cadquery.occ_impl.exporters' has no attribute 'utils'
```

bragostin, 2025-01-25:
Having the same issue after a pip update I commented out `#reload(cq.occ_impl.exporters.utils)` in `cq_utils.py` as a workaround and now cq-editor works again for me.

jmwright, 2025-01-28:
@jdegenstein @bragostin Please confirm that #470 fixes the issue.

jmwright, 2025-01-28:
Nevermind, I had to use #470 to help fix the tests anyway and it worked fine.

closed 2025-01-28 by jmwright

bragostin, 2025-01-28:
Yes it works thank you!