CadQuery/CQ-editor #472: Looking for static build testers on macos-x86_64 and macos-arm64

jdegenstein, 2025-01-29:
Hello,

I have maintained a fork of CQ-editor at https://github.com/jdegenstein/jmwright-CQ-Editor for the past several years.

In that time I have released static builds of CQ-editor built with pyinstaller that bundle CadQuery, build123d and other useful packages. These builds provide a "download, unzip and run" user experience for python / CQ newcomers. These builds are also run on a weekly basis via github actions.

https://github.com/jdegenstein/jmwright-CQ-Editor/releases

I recently finished my effort to also release static builds for macos-arm64 (and have long released macos-x86_64) but I could use some testers on these platforms. Bonus points if any advice can be offered related to eventually signing these packages for these platforms or other related QoL improvements for macos.

Thank you

jmwright, 2025-01-29:
@michaelgale Would you be able to help test the MacOS builds? We're talking about consolidating builds into this main repo and would like to do some testing on the PyInstaller builds before going so. I know you have PR #111 still open, which I think is related to this.

evrenesat, 2025-01-29:
Hi, JFY, I just found CadQuery and tried to run macos-arm64 build on macos 15.1, but OS didn't allow me to run it. I entered my password many many times, allowed many libs in preferences, but couldn't reach end of prompts and finally gave up!

[image]

jmwright, 2025-01-29:
There are a lot of .so files bundled, so if you are having to confirm every one of them without a "do the same with the rest" checkbox, you will be there for quite some time. We need to come up with a way to get MacOS packages signed.

jdegenstein, 2025-01-29:
Some relevant discussion here about a possible workaround:

https://github.com/jdegenstein/jmwright-CQ-Editor/issues/20

Would love if this could be packaged in a script somehow and ran by the end user with elevated permissions. Can anyone test this workaround and then maybe we can collaborate on creating a script and necessary instructions?

michaelgale, 2025-01-31:
> [@michaelgale](https://github.com/michaelgale) Would you be able to help test the MacOS builds? We're talking about consolidating builds into this main repo and would like to do some testing on the [...]

Sure, I'd be happy to help if I can.  I do have several MacOS machines (both Intel and arm64 'M') if necessary to test both.  It was a while ago in which I built CQ-editor app bundles (using pyinstaller etc.) and do remember it was a pain.  Building software bundles on macOS is so heavily coupled to the Xcode ecosystem and ensuring consistency of dependencies (which are architecture and macOS version dependent!) is tedious and time-consuming.

jmwright, 2025-01-31:
Thanks @michaelgale . If you could test the x86_64 and ARM64 MacOS builds [here](https://github.com/jdegenstein/jmwright-CQ-Editor/releases) and give feedback, that would be great.

michaelgale, 2025-01-31:
Just did a quick check of CQ-Editor 0.30-dev-7 for macOS arm64.  As expected, the usual macOS security warnings for unsigned binaries occurred.  However, since all the bundled libraries for this app are dynamically loaded from the app's `_internal` folder; every single dylib triggers a security trap/exemption--obviously not workable for an end user!  I tried stripping the quarantine extended attributes for entire shared library path:

```shell
$ cd CQ-editor-MacOS-arm64/CQ-editor
$ xattr -rd com.apple.quarantine ./_internal
```

However, it still doesn't work:

```shell
$ ./CQ-editor.sh
Traceback (most recent call last):
  File "run.py", line 12, in <module>
  File "PyInstaller/loader/pyimod02_importers.py", line 384, in exec_module
  File "cq_editor/__main__.py", line 12, in <module>
  File "PyInstaller/loader/pyimod02_importers.py", line 384, in exec_module
  File "cq_editor/main_window.py", line 4, in <module>
  File "PyInstaller/loader/pyimod02_importers.py", line 384, in exec_module
  File "logbook/__init__.py", line 14, in <module>
  File "PyInstaller/loader/pyimod02_importers.py", line 384, in exec_module
  File "logbook/base.py", line 1165, in <module>
  File "PyInstaller/loader/pyimod02_importers.py", line 384, in exec_module
  File "logbook/handlers.py", line 17, in <module>
  File "PyInstaller/loader/pyimod02_importers.py", line 384, in exec_module
  File "ssl.py", line 100, in <module>
ImportError: dlopen(/Users/michaelgale/tmp/CQ-editor-MacOS-arm64/CQ-editor/_internal/lib-dynload/_ssl.cpython-311-darwin.so, 0x0002): Symbol not found: _X509_STORE_get1_objects
  Referenced from: <CBD20FE3-6EF6-3B93-8EEF-E38E0ECF6534> /Users/michaelgale/tmp/CQ-editor-MacOS-arm64/CQ-editor/_internal/lib-dynload/_ssl.cpython-311-darwin.so
  Expected in:     <8DC5F7BD-4DE8-3B84-B5D7-567D58C8A21E> /Users/michaelgale/tmp/CQ-editor-MacOS-arm64/CQ-editor/_internal/libcrypto.3.dylib
[PYI-41105:ERROR] Failed to execute script 'run' due to unhandled exception!
```

I wonder if it might be worth building an app bundle using pyinstaller's "one file" option.  I presume this would statically link all of the underlying dependencies into a single binary which would be easier to bypass and authorize for end users.  The downside is of course the size of the resulting app bundle would be huge.  I vaguely remember nearly 1 GB app bundles popping out of pyinstaller!  Newer versions of pyinstaller might be better/more efficient.

adam-urbanczyk, 2025-01-31:
I'm not going to work on this project anymore, so pick your poison, but using conda constructor is much saner and already working. See here: https://github.com/CadQuery/CQ-editor/releases/tag/nightly and here: https://github.com/CadQuery/CQ-editor/blob/master/conda/construct.yaml and here  https://github.com/CadQuery/conda-packages/blob/master/constructor-build.yml