CadQuery/CQ-editor #482: ValueError: offset() 'Invalid mode: r'  in sketch example

gitterman, 2025-02-09:
I tried the examples on https://cadquery.readthedocs.io/en/latest/sketch.html#sketches-spanning-multiple-elements and get the error 'Invalid mode: r' when I specify this mode.
The offset() page mentions  "mode (Literal['a', 's', 'i', 'c', 'r'])" but there is no explanation on these and mode 'r' doesn't seem to exist.

adam-urbanczyk, 2025-02-09:
I can reproduce this issue locally. No clue what is the rootcause:

```Output folder: d:\cq-editor2\Lib
Extract: _nsis.py
Extract: _system_path.py
Output folder: d:\cq-editor2
Extract: LICENSE
By continuing this installation you are accepting this license agreement:
d:\cq-editor2\LICENSE
Please run the installer in GUI mode to read the details.

CQ-editor will now be installed into this location:
d:\cq-editor2

Unpacking payload...
Output folder: d:\cq-editor2\conda-meta
Extract: history
Output folder: d:\cq-editor2
Extract: _conda.exe
Extract: pre_uninstall.bat
Output folder: d:\cq-editor2
Output folder: d:\cq-editor2\pkgs
Extract: urls
Extract: urls.txt
Extract: post_install.bat
Output folder: d:\cq-editor2\pkgs\cache
Extract: 09cdf8bf.json
Extract: 5afe41e9.json
Extract: 6219f36a.json
Extract: 8ff15dc5.json
Output folder: d:\cq-editor2\pkgs
Output folder: d:\cq-editor2\pkgs\aiohappyeyeballs-2.4.4-pyhd8ed1ab_1\info
Extract: repodata_record.json
Output folder: d:\cq-editor2\pkgs\aiohttp-3.11.12-py312h31fea79_0\info
Extract: repodata_record.json
Output folder: d:\cq-editor2\pkgs\aiosignal-1.3.2-pyhd8ed1ab_0\info
Extract: repodata_record.json
Output folder: d:\cq-editor2\pkgs\alabaster-1.0.0-pyhd8ed1ab_1\info
Extract: repodata_record.json
Output folder: d:\cq-editor2\pkgs\ampl-asl-1.0.0-he0c23c2_2\info
Extract: repodata_record.json
Output folder: d:\cq-editor2\pkgs\aom-3.9.1-he0c23c2_0\info
Extract: repodata_record.json
Output folder: d:\cq-editor2\pkgs\arrow-1.3.0-pyhd8ed1ab_1\info
Extract: repodata_record.json
Output folder: d:\cq-editor2\pkgs\astroid-3.3.8-py312h2e8e312_0\info
Extract: repodata_record.json
Output folder: d:\cq-editor2\pkgs\asttokens-3.0.0-pyhd8ed1ab_1\info
Extract: repodata_record.json
Output folder: d:\cq-editor2\pkgs\atomicwrites-1.4.1-pyhd8ed1ab_1\info
Extract: repodata_record.json
Output folder: d:\cq-editor2\pkgs\attrs-25.1.0-pyh71513ae_0\info
Extract: repodata_record.json
Output folder: d:\cq-editor2\pkgs\autopep8-2.0.4-pyhd8ed1ab_0\info
Extract: repodata_record.json
Output folder: d:\cq-editor2\pkgs\babel-2.17.0-pyhd8ed1ab_0\info
Extract: repodata_record.json
Output folder: d:\cq-editor2\pkgs\backports-1.0-pyhd8ed1ab_5\info
Extract: repodata_record.json
Output folder: d:\cq-editor2\pkgs\backports.tarfile-1.2.0-pyhd8ed1ab_1\info
Extract: repodata_record.json
Output folder: d:\cq-editor2\pkgs\bcrypt-4.2.1-py312h2615798_0\info
Extract: repodata_record.json
Output folder: d:\cq-editor2\pkgs\beautifulsoup4-4.13.3-pyha770c72_0\info
Extract: repodata_record.json
Output folder: d:\cq-editor2\pkgs\binaryornot-0.4.4-pyhd8ed1ab_2\info
Extract: repodata_record.json
Output folder: d:\cq-editor2\pkgs\black-25.1.0-py312h2e8e312_0\info
Extract: repodata_record.json
Output folder: d:\cq-editor2\pkgs\bleach-6.2.0-pyh29332c3_4\info
[... 1671 lines cut ...]
  wcwidth            conda-forge/noarch::wcwidth-0.2.13-pyhd8ed1ab_1 
  webencodings       conda-forge/noarch::webencodings-0.5.1-pyhd8ed1ab_3 
  whatthepatch       conda-forge/noarch::whatthepatch-1.0.7-pyhd8ed1ab_1 
  wheel              conda-forge/noarch::wheel-0.45.1-pyhd8ed1ab_1 
  win32_setctime     conda-forge/noarch::win32_setctime-1.2.0-pyhd8ed1ab_0 
  win_inet_pton      conda-forge/noarch::win_inet_pton-1.1.0-pyh7428d3b_8 
  wslink             conda-forge/noarch::wslink-2.2.2-pyhd8ed1ab_0 
  x264               conda-forge/win-64::x264-1!164.3095-h8ffe710_2 
  x265               conda-forge/win-64::x265-3.5-h2d74725_3 
  yaml               conda-forge/win-64::yaml-0.2.5-h8ffe710_2 
  yaml-cpp           conda-forge/win-64::yaml-cpp-0.8.0-h63175ca_0 
  yapf               conda-forge/noarch::yapf-0.43.0-pyhd8ed1ab_1 
  yarl               conda-forge/win-64::yarl-1.18.3-py312h31fea79_1 
  zeromq             conda-forge/win-64::zeromq-4.3.5-ha9f60a1_7 
  zipp               conda-forge/noarch::zipp-3.21.0-pyhd8ed1ab_1 
  zlib               conda-forge/win-64::zlib-1.3.1-h2466b09_2 
  zstandard          conda-forge/win-64::zstandard-0.23.0-py312h7606c53_1 
  zstd               conda-forge/win-64::zstd-1.5.6-h0ea2cb4_0 

Downloading and Extracting Packages: ...working... done
Preparing transaction: - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ done
Executing transaction: / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ | / - \ |Ō䴀晶
::error:: Failed to link extracted packages to d:\cq-editor2!

```

jmwright, 2025-02-09:
@adam-urbanczyk Based on that issue, it seems like this problem is outside of our control for now.

@gitterman Is this blocking you from using CadQuery? If so, and if you can install the cadquery package, you can add the following code to your CadQuery script to display your models without needing an external viewer.

```python
from cadquery.vis import show

show(my_cq_object)
```

There are also other ways to install CQ-editor depending on your use of virtual environments and whether you have git installed.

gitterman, 2025-02-09:
Thanks for the hint.
I have a version of cq editor running, which I installed last year using conda.
The current installation description unfortunately goes another way.
For now, I am happy with my version, although some features obviously don't work with it.

adam-urbanczyk, 2025-02-09:
I was able to install using headless mode:
```
cmd /C START /WAIT CQ-editor-master-Windows-x86_64(2).exe /S /NoRegistry=1 /AddToPath=0 /D=D:\CQ-editor3
```
But I need to start via `Scripts\CQ-editor.exe`, i.e. `run.bat` does not seem to be installed.

[image]

gitterman, 2025-03-11:
I regard this as closed, although the offset entry in the documentation https://cadquery.readthedocs.io/en/latest/classreference.html#cadquery.Sketch.offset still does not tell, what the different modes stand for.

closed 2025-03-11 by gitterman

nobkd, 2025-03-11:
https://cadquery.readthedocs.io/en/latest/_modules/cadquery/sketch.html

```py
Modes = Literal["a", "s", "i", "c", "r"]  # add, subtract, intersect, construct, replace
```