CadQuery/CQ-editor #493: Installing extra packages of wiki might be outdated

LokiLin14, 2025-02-27:
I am following the quickstart of the wiki. 

```md
Installing extra packages

mamba, and pip are bundled with the CQ-editor installer and available for package installation.

First activate the environment, then call mamba or pip to install additional packages.

On windows.:

C:\Users\<username>\cq-editor\Scripts\activate
mamba install <packagename>

On Linux/MacOS.:

source $HOME/cq-editor/bin/activate
mamba install <packagename>
```

but no activate files are in bin: 
```bash
$ find -name "activate"
./lib/python3.12/venv/scripts/common/activate
``` 

is there a new way to activate the pip / conda environments to install additional packages?

I noticed there is a `_conda` in the installed directory but running `_conda activate ./` errors with `CondaError: Run 'conda init' before 'conda activate'`.

jmwright, 2025-02-27:
@LokiLin14 What wiki are you referring to? I show [this repo's wiki](https://github.com/CadQuery/CQ-editor/wiki) being empty.

LokiLin14, 2025-02-27:
Oops, sorry, by wiki, I meant the [docs](https://cadquery.readthedocs.io/en/latest/installation.html#installing-extra-packages).

jmwright, 2025-02-27:
@LokiLin14 Do you need to use the installer, or can you use [conda](https://github.com/CadQuery/CQ-editor?tab=readme-ov-file#installation-condamamba) or [pip](https://github.com/CadQuery/CQ-editor?tab=readme-ov-file#installation-pip) directly?

LokiLin14, 2025-02-27:
oh nice, using the conda installation worked

i tried using the pip installation before but when I tried to run it, it errored `X Error of failed request:  BadWindow (invalid Window parameter)` , so I didn't try the conda install but the conda install runs without issue.

jmwright, 2025-02-27:
The pip installation is a newer option, so there has not been time to work out issues with it. For reference, which Linux distro/version do you run?

LokiLin14, 2025-02-27:
I am running EndeavorOS using the Wayland KDE Desktop. I haven't tweaked any of the compositor / KDE settings just running the default configuration that the installer sets up.

jmwright, 2025-02-28:
@LokiLin14 If you are willing to try installing via pip in another environment again, you could try [this fix](https://github.com/CadQuery/CQ-editor/issues/266#issuecomment-2113470312) for Wayland. If it works for you, I'll add it to the docs as a troubleshooting measure.

jmwright, 2025-02-28:
@LokiLin14 Note that issue has a capitalization error. You will need to run `CQ-editor --platform xcb`.

jmwright, 2025-02-28:
Added to the readme in #497 497

closed 2025-02-28 by jmwright