CadQuery/CQ-editor #494: Rerunning a script with `import scipy.optimize` results in import errors

LokiLin14, 2025-02-27:
Running the following script by clicking "Render" will succeed. 

```python
import scipy.optimize
```

But clicking the "Render" button again will fail with the error: `ModuleNotFoundError: No module named 'scipy.optimize._highspy.simplex_constraints'; 'scipy.optimize._highspy._core' is not a package`

### Things I tried  

#### First import is real
The first Render seems to succeed in importing the scipy.optimize module because I can use scipy.optimize: I run the following script
```python
import numpy as np
from scipy.optimize import minimize

# Define a simple quadratic function
def objective_function(x):
    return x**2 + 4 * x + 4

# Initial guess
initial_guess = np.array([0])

# Perform the minimization
result = minimize(objective_function, initial_guess)

# Print the result
print("Optimal value of x:", result.x)
print("Minimum value of the function:", result.fun)
```

It correctly prints: 
```
Optimal value of x: [-2.00000002]
Minimum value of the function: 0.0
```

Clicking "Render" again results in the same error `ModuleNotFoundError: No module named 'scipy.optimize._highspy.simplex_constraints'; 'scipy.optimize._highspy._core' is not a package`

#### Local scope doesn't seem to help 

I thought that maybe the previous import of scipy could be causing issues when scipy is re-imported so I imported inside a local scope. 

```python
def import_scipy(): 
    import scipy.optimize
import_scipy()
```

Rendering this twice still causes the same module not found error. 

#### Environment details

`conda list` shows
```
cq-editor                 master          0.4.0_11_g7e85b87    cadquery
scipy                     1.15.2          py312ha707e6e_0    conda-forge
```

I am running the cq-query installed via `conda install -c cadquery -c conda-forge cq-editor=master`. The run command is `python -m cq_editor
`.

jmwright, 2025-02-27:
Is the "Reload imported modules" setting set in the preferences dialog?

LokiLin14, 2025-02-27:
yes, "reload imported modules" was set. Turning it off seems to have fixed the issue.

closed 2025-02-27 by jmwright