CadQuery/CQ-editor #504: Shapes from DXF brings wrong results when cutting them

DeBalog, 2025-03-14:
An inner shape was created, as well as an outer one, both in Inkscape.
The goal is to cut the inner shape out of the outer one. Unfortunately, I wasn't able to achieve this—see the code and the pictures below.

In both test cases, I expect a border that flows around the figure.

```python
import cadquery as cq

def moveToOrigo (shape):
    vectorToCenter = shape.val().Center()
    return shape.translate(- vectorToCenter)
cq.Workplane.moveToOrigo = moveToOrigo

def dxf2shape(dxf, z = 1):
    dxfShape = cq.importers.importDXF(dxf).wires().toPending().extrude(z)
    return dxfShape.moveToOrigo ()

inner = dxf2shape ("mann2.dxf")
outer = dxf2shape ("mann2Rand.dxf")
innerZ5 = dxf2shape ("mann2.dxf", 5)

# Testing equal thickness
show_object (inner)
show_object (outer)
show_object (inner.          translate([30, 0]))
show_object (outer.          translate([60, 0]))
show_object ((outer - inner).translate([90, 0]))

# testing unequal thickness
show_object (innerZ5.          translate([0,  -50]))
show_object (outer.            translate([0,  -50]))
show_object (innerZ5.          translate([30, -50]))
show_object (outer.            translate([60, -50]))
show_object ((outer - innerZ5).translate([90, -50]))
```

[image]

The top and bottom faces are missing in both cutted figures on the right. In the bottom line the cutted figure has a wrong bottom face (it is to deep) and the inner walls are to high.

[mann2.zip](https://github.com/user-attachments/files/19252136/mann2.zip)

[mann2Rand.zip](https://github.com/user-attachments/files/19252155/mann2Rand.zip)

I am using:

[image]

jmwright, 2025-03-17:
Please try displaying with the in-built show functionality. If you still have the issue, it's a CadQuery question instead of a CQ-editor question.

```python
from cadquery.vis import show
...
show(my_model)
```

DeBalog, 2025-03-24:
I changed the code, as @jmwright recommended and the error persists, so this is a CadQuery question.

[image]