CadQuery/CQ-editor #506: sketch offset mode 'r' doesn't work

gitterman, 2025-03-18:
I have installed miniforge and cloned CQ editor from here last week and tried the last example on https://cadquery.readthedocs.io/en/latest/sketch.html but mode 'r' is not working.

These were my steps:
```
curl -L -o miniforge.exe https://github.com/conda-forge/miniforge/releases/latest/download/Miniforge3-Windows-x86_64.exe
miniforge.exe /InstallationType=JustMe /RegisterPython=0 /NoRegistry=1 /NoScripts=0 /S /D=%USERPROFILE%\Miniforge
 git clone https://github.com/CadQuery/CQ-editor.git
 "%USERPROFILE%/Miniforge/Scripts/activate.bat" "%USERPROFILE%/Miniforge"
 mamba env create -f cqgui_env.yml -n cqgui
 conda clean --all
 cd "%USERPROFILE%\Miniforge\CQ-Editor"
 mamba activate cqgui
 python run.py
```

This is the sketch and the output is shown on the attached picture.
```
import cadquery as cq

sketch  = (cq.Sketch()
.rect(1.0, 4.0)
.circle(1.0)
.clean()
)
sketch_offset = sketch.copy().wires().offset(-0.25, mode='r')
result = cq.Workplane("front").placeSketch(sketch).extrude(1.0)
result = result.faces(">Z").workplane().placeSketch(sketch_offset).cutBlind(-0.50)
show_object( result ) 
```
[image]

jmwright, 2025-03-18:
I cannot reproduce this issue.

Can you run `conda list` and look at the CadQuery version? You will need either version 2.5.2 or master. You can also run the following, but it may not be as illuminating as the conda list output.
```python
import cadquery as cq
print(cq.__version__)
```

gitterman, 2025-03-18:
conda list gives me
` cadquery                  2.4.0           2.4.0_0_gc44978d    CadQuery`
So this explains the issue.
What it does NOT explain though, is why my installation doesn't have the latest rev, as I have installed miniforge as well as cq-editor from the latest version of last week.
I did not install cadquery separately before though, as I thought, that cq-editor brings all the stuff with it.
Am I missing something here, or is the cadquery rev. from the installed packages not current?

closed 2025-03-19 by gitterman