CadQuery/CQ-editor #508: bug: editor cursor jumps to begin of file when saving file

ilka-schulz, 2025-04-12:
When pressing `Ctrl+S` or selecting `File -> Save` from the main menu, the cursor in the text editor on the left jumps to the very beginning of the currently viewed (and now saved) file.

Fresh install on Debian 12 (x86_64), Python 3.11.2, all default settings.

jmwright, 2025-04-12:
How did you install CQ-editor and can you verify which version you are running? This bug was fixed in #491 . I run Debian 12 and do not have the issue any longer in version 0.5.

Please also verify that you are not running a fork of CQ-editor. There are a few of those around that are not synced with this main branch.

If you verify those things and you still have the issue, could you post a screenshot of the Editor section of your Preferences dialog?

ilka-schulz, 2025-04-12:
@jmwright thanks for the quick response!

I was indeed running CQ editor `0.4.0`, installed via `pip3 install cq-editor`. I just updated to the new `0.5.0` version released a couple days ago and it appears to implement a nice workaround for the issue. Closing the issue.

closed 2025-04-12 by ilka-schulz