CadQuery/CQ-editor #509: regression in version 0.5.0: Shortcut Shift+F1 does not work anymore

ilka-schulz, 2025-04-12:
I just upgraded from version `0.4.0` to `0.5.0` under Debian 12 (x86_64), Python 3.11.2, installed via `pip3 install cq-editor`.

The  shortcut `Shift+F1` (also available via main menu -> `View` -> `Fit`) does no longer work (no effect). Selecting the function from the main menu does, however, still work.

jmwright, 2025-04-12:
It works for me if I have the 3D viewer selected before pressing the key combination. The editor seems to be intercepting the key combo when it has focus. Can you verify that it works for you if you click in the 3D viewer first?

ilka-schulz, 2025-04-13:
Honestly, I just started CQ editor again (still the exact same setup) and I can not reproduce the bug at all. Regardless of whether I focus the editor or the 3D viewer, `Shift + F1` works flawlessly.

I am on vacation this week until easter and promised myself to spend as much time as possible away from my computer but I am planning to investigate into this issue later. If I am able to reproduce the issue, I am looking forward to submit a PR.

ilka-schulz, 2025-04-24:
I used CQ editor a lot during the past two days and I can not reproduce the bug at all. I might consider closing the issue but maybe so. else might reproduce it.