CadQuery/CQ-editor #514: editor / log viewer etc. panels disappear: How to get them back?

bkpgh, 2025-04-25:
I often have difficulty retrieving panels when they disappear due to some accidental keyboard or mouse action (I suppose). Sometimes, they are detached from the main window and I don't know how to put them back where they were. Dragging them sort of works, but they end up with funny sizes and don't integrate with tabs.  For example, suppose the editor panel disappears. Selecting View/Editor does nothing as far as I can tell. How does one gracefully manipulate the different panels.

It appears I'm running 0.3.0dev.

Thanks

jmwright, 2025-04-26:
I cannot reproduce on CQ-editor 0.5. I can undock and redock panels, and toggle their visibility.

jmwright, 2025-04-29:
I tried running CQ-editor 0.5 in the background (`CQ-editor &`) on Debian 12 and cannot reproduce. The worst thing that happens is that the panel ends up wider or narrower than it was when I re-dock it, requiring me to drag the divider a bit.