CadQuery/CQ-editor #515: Mouse doesn't work in Editor

pefau72, 2025-05-08:
After upgrading to latest CQ-editor on Mac, my mouse doesn't work in the editor window. Works fine in all other windows.

jmwright, 2025-05-08:
Can your elaborate on what "doesn't work" means? By latest version do you mean 0.5, or did you install from git? Did you upgrade via conda or pip? More information is needed.

jmwright, 2025-05-08:
Following the bug report steps outlined in the readme will help diagnose this, especially the steps to reproduce and info on exactly what about the Editor/mouse is not working.

https://github.com/CadQuery/CQ-editor?tab=readme-ov-file#reporting-a-bug

Also, does the problem exist if you install CQ-editor into a clean environment?

pefau72, 2025-05-09:
> jmwright

It is version 0.6.dev0
I installed https://cadquery.readthedocs.io/en/latest/installation.html (not Git)
I used conda
I cant click in the editor with the mouse.

jmwright, 2025-05-09:
I am not able to reproduce, and the only changes that have been made to this repo since the 0.5 release are to the documentation, the version number, and the launcher script for the pre-built conda packages. Nothing related to the editor. I suspect that conda broke CQ-editor for you during the upgrade. Can you try a fresh install in a new conda environment?