CadQuery/CQ-editor #520: Disable Toggle Comment when Editor is not visible

lorenzncode, 2025-06-02:
There is a new `Toggle Comment` item under the Edit menu and shortcut key.  I'd suggest to disable the feature when the Editor widget is not visible for users of external editors.

jmwright, 2025-06-03:
@lorenzncode Is there a functional reason to do this (i.e. it causes a problem), or is it more of a user experience reason (i e. It causes confusion)?

lorenzncode, 2025-06-03:
@jmwright It's more for user experience (nothing major).   Since I'm not using the editor I'd prefer only the viewer options to be presented or enabled.  Then I can safely ignore editor features and no chance I might comment something out that is not visible.  I guess it would only comment line #1 if I were to trigger it accidentally.

lorenzncode, 2025-08-20:
Thanks @jmwright !   In my case I don't toggle the Editor visibility. The Editor visibility is always off based on my existing preferences. I pass in the Python input file from CLI.